cujomalainey/antplus-arduino #24: Yun Usb

fooldome, 2019-11-13:
Hi,
Do you know if it's possible to get this library to build on a Arduino Yun?
With it's spare standard usb port I'd like to try and implement direct reading of a Ant+ usb receiver for multiple heart rate monitors, but initial tests seems the library is incompatible with Yun. Can you provide any help or insight?

cujomalainey, 2019-11-13:
It should be compatible since it only depends on a serial port, what kind of errors are you seeing?

cujomalainey, 2019-11-13:
also, if you are using the arduino ide, did you download ant-arduino as well? You need it drive the underlying serial protocol to the radio that antplus-arduino builds on

cujomalainey, 2019-11-19:
It appears the USB port is behind the AR9331 which is design to be host to usb serial flash, not a fully featured USB host, therefore this is likely impossible on the Yun.

closed 2019-11-19 by cujomalainey

fooldome, 2019-11-19:
Thanks for the comments. I eventually fixed the build issues only to slowly come to the same conclusion that a direct read of an Ant+ usb may not be possible. That said there seems to be a way forward using bridge processes for the linux side of the Yun that can do it, so I'm currently trying to port a successful python test I did on Raspberry Pi.

cujomalainey, 2019-11-19:
Good luck