cujomalainey/antplus-arduino #29: Will this work with the ESP32?

bsperryn, 2021-04-09:
Hi, I've read that the ant arduino only with this the NRF51/52, but this library is said to work with all architectures, so does it work with the ESP32?

cujomalainey, 2021-04-09:
Hi @bsperryn, 

Yes you can use this with an ESP32 with an external nRF52/51. Dynastream only supports [nordic chips](https://www.thisisant.com/developer/components/nrf52832/) to run their radio firmware so you have to have a nRF52 somewhere to actually use ANT. The reason all archs are supported is because nRF chips directly from Dynastream come loaded with a serial/spi interfacing driver (called the network processor) so you can communicate over serial with the radio. (see in ant-arduino [*SerialAnt](https://github.com/cujomalainey/ant-arduino/blob/master/src/BaseClasses/ANT_BaseSerialAnt.h#L7) vs [*NativeAnt](https://github.com/cujomalainey/ant-arduino/blob/master/src/BaseClasses/ANT_BaseNativeAnt.cpp#L36) (ignore the comments in NativeAnt header, i realized it is a copy paste error) Typically in my testing I use a [Huzzah32](https://www.adafruit.com/product/3405) (aka esp32) with this [nifty board](https://github.com/reeganm/ANT-FeatherWing) an old roommate made for me with the ArduinoSerialAnt driver from ant-arduino. If you are not using the feather ecosystem and are just breadboarding you can grab a [D52 dev](https://www.digikey.ca/en/products/detail/garmin-canada-inc/D52QD2M6IA-A-TRAY/6149453) module (an nRF52 just preloaded with ANT) and just [wire your communication pins](https://github.com/cujomalainey/ant-arduino/wiki/Developer's-Guide#using-pinouts-on-d52qd2m6ia-a-tray) directly. Another option (with a [bit of fiddling](http://orrmany.hu/wiki/doku.php?id=arduino:s340_softdevice_adafruit_nrf52840_feather_express) around to replace the radio firmware) you can use something like the [Feather nRF52840 Express](https://www.adafruit.com/product/4062) and use ArduinoNativeAnt and use ANT via the onboard radio firmware.

Hope that clears things up, let me know if that answered everything or if you have anymore questions.

bsperryn, 2021-04-11:
Thank you! Would I also be able to use the BLE at the same time with the d52 dev or separate board?

cujomalainey, 2021-04-12:
If you use the ESP32 you can use the ESP's BLE. If you program the nRF52 directly you can load  firmwares  (s340) that support BLE+ANT.

cujomalainey, 2021-04-18:
If you have any questions feel free to post to the google group or reopen this bug

closed 2021-04-18 by cujomalainey

hayschan, 2023-08-29:
I am familiar with the ESP-IDF framework, and the ESP32-S3 and ESP32-C3 MCUs. Planning to use them to build some open source ANT+ sensors for cycling, e.g. cadence, speed, lights.

Does this library support ESP-IDF?

cujomalainey, 2023-08-29:
@hayschan not currently but it should be very easy to implement. You just need to add a [shim](https://github.com/cujomalainey/ant-arduino/tree/master/src/MainClasses) for what type of port you are using.

E.g. [Arduino serial](https://github.com/cujomalainey/ant-arduino/blob/master/src/MainClasses/ANT_ArduinoSerialAnt.cpp) is less than 40 lines of code. If you want to use antplus-arduino you will also need to add the [callbacks class](https://github.com/cujomalainey/ant-arduino/blob/master/src/MainClasses/ANT_ArduinoSerialAntWithCallbacks.cpp) another like 5 lines of code.

I tried esp-idf a long time ago but couldnt figure out the pin mapping and got distracted by something else, if you are willing to send PRs I will happily review and integrate them.

hayschan, 2023-08-30:
What is the relationship between [antplus-arduino](https://github.com/cujomalainey/antplus-arduino) and [ant-arduino](https://github.com/cujomalainey/ant-arduino)?

> if you are willing to send PRs I will happily review and integrate them.

Yeah sure.

Fmstrat, 2025-05-09:
I know this was closed a long time ago, but I have a similar use case. Could this be integrated into ESPHome? We have a temperature sensor in our garage where we have the bike trainer, and would like the broadcast the ambient temperature from it via ANT+ to our Garmin watches.

cujomalainey, 2025-05-11:
> I know this was closed a long time ago, but I have a similar use case. Could this be integrated into ESPHome? We have a temperature sensor in our garage where we have the bike trainer, and would like the broadcast the ambient temperature from it via ANT+ to our Garmin watches.

Hi @Fmstrat 

This likely cannot be easily integrated into esphome. It's possible just not easy. You would need to wrap my code with an esphome driver. Won't be too bad if you hard code it to what you are needing. But if you want to make it generic and upstreamable then it will be much harder as you will have to generate a lot of yaml characteristics. Do you need to use esphome? If not it would likely be easiest to just use the Arduino driver directly for the sensor and plug the values into the environment profile.