cujomalainey/antplus-arduino #30: ANT+ Examples do not build on Arduino for Adafriut nRF52840 Express

jludwig75, 2021-11-04:
I've installed the board libraries for the Adafruit Feather nRF52840 Express and ANT-Arduino and ANTPLU-Arduino. I get errors trying to build the HeartRateMonitor example. The code uses ArduinoSerialAntWithCallbacks, but I believe I should use ArduinoNativeAntWithCallbacks with this board. Is that correct? Are there instructions on how to build native ANT code for this board?

cujomalainey, 2021-11-04:
Hi @jludwig75 

Try adding `#include "HardwareSerial.h` above the top of your sketch

Are you trying to use on nrf52 locally or an external nrf52? The native is for on chip, the serial for external chips such as the D52

cujomalainey, 2021-11-04:
Also since its a build issue, if you can share how you are building the sketch (e.g. if by platformio the ini file) and i can quickly reproduce this

jludwig75, 2021-11-04:
I am trying to build to run on the nrf52. I first tried using platformio and then the Arduino IDE to make sure there wasn't some difference. Let me upload what I have.

jludwig75, 2021-11-05:
This is what is in platformio.ini:

[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
lib_deps = 
	cujomalainey/ANT@^2.0.1
	cujomalainey/ANT+@^1.1.1

jludwig75, 2021-11-05:
Here is my project:

[arduino-ant-test.tar.gz](https://github.com/cujomalainey/antplus-arduino/files/7479550/arduino-ant-test.tar.gz)

But is it using the serial interface. I'll upload my best guess at using the native interface.

Thank you for looking at this. I'd really like to use your library to create a replacement sensor for my indoor cycle trainer.

jludwig75, 2021-11-05:
This is my best attempt at building native. I think it is pretty naive. I also know I need to get the softdevice running. I'm nit sure how to do that. You can tell I don't know much yet.

[arduino-ant-test-native.tar.gz](https://github.com/cujomalainey/antplus-arduino/files/7479582/arduino-ant-test-native.tar.gz)

jludwig75, 2021-11-05:
Including HardwareSerial.h doesn't help. It looks like the nrf52 core does not define Serial for nrf52840, just Serial1. It does define both for nRF52832. I just changed everything to Serial1. I also added the soft device headers as a plarformio lib to my project and made some include changes to pull in the right includes. I am now down to just these errors:

.pio/build/adafruit_feather_nrf52840/libFrameworkArduino.a(Uart.cpp.o): In function `serialEventRun()':
Uart.cpp:(.text._Z14serialEventRunv+0x8): undefined reference to `Adafruit_USBD_CDC::available()'
Uart.cpp:(.text._Z14serialEventRunv+0x2c): undefined reference to `Serial'
collect2: error: ld returned 1 exit status
*** [.pio/build/adafruit_feather_nrf52840/firmware.elf] Error 1

This is my latest project: 
[arduino-ant-test-native.tar.gz](https://github.com/cujomalainey/antplus-arduino/files/7479701/arduino-ant-test-native.tar.gz)

jludwig75, 2021-11-05:
Are these good instructions for building the boot loader with the soft device and installing it on the board?

https://blogarak.wordpress.com/2020/03/15/s340-softdevice-adafruit-nrf52840-express-feather/

Sorry if I'm asking outside of the scope of this project, but I think it would be great to make this work. I would be very happy to contribute back samples and documentation.

cujomalainey, 2021-11-05:
Yep, those are good for the S340 example, but they are definitely a bit much, but I figured out this issue, it looks like its a bug in Adafruits bsp package. You need to include `#include <bluefruit.h>` at the top of your sketch in order to get the reference to the `Serial` object.

see https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/custom-hrm

Let me know if that helps. Thanks

cujomalainey, 2021-11-09:
closing as there has been no response for a few days, feel free to open if it did not resolve the problem

closed 2021-11-09 by cujomalainey