cujomalainey/antplus-arduino #33: Use library on Seeeduino Xiao ble sense

ScrapTech, 2022-07-29:
Hi

I am trying to make an antplus sensor to send data to my garmin edge.
For this I was going to use the Seeed Xiao BLE sense board (nRF52840) together with the antplus-arduino library.

I installed the Seeed Xiao, the antplus library and ant library on the arduino IDE as described in following link: [Getting Started with XIAO BLE (Sense) - Seeed Wiki 1](https://wiki.seeedstudio.com/XIAO_BLE/)
Now I can upload example projects from seeeduino to the board and I can compile examples from the antplus library on other boards.

Whenever I try to compile an example from the antplus library on the seeeduino board, I get following errors:

[image]

When I look in the header files of the seeeduino board, I noticed indeed that the Serial was defined as UART1:
[image]
It seems that the antplus library can not handle this, and does not recognize UART1 as a type?

Thanks in advance for any help!

cujomalainey, 2022-07-31:
Hi, I'll take a look into this, looks like a minor config issue. Are you using Arduino or mbed as your primary OS?

ScrapTech, 2022-08-02:
Thank you! I am using Arduino as my primary OS

cujomalainey, 2022-08-02:
Figured out the issue, its because the board is using mbed under the arduino HAL, so the system is having type collisions. I'll push a fix later tonight.

Also are you planning to use ANT on the Seeed chip or an external D52 chip?

ScrapTech, 2022-08-02:
I'm planning to use the seeed chip itself, not an external one.

cujomalainey, 2022-08-03:
Got it, best of luck with flashing the softdevice, i have seen mixed results. I should have a test fix for you in a few hours.

cujomalainey, 2022-08-03:
I pushed a fix to ant-arduino, can you test it out?

https://github.com/cujomalainey/ant-arduino/commit/afcbca69050a3019ded8e7b121e1b55b35006a80

cujomalainey, 2022-08-04:
Sounds good, I'll cut a micro release once I fix the gcc complaints about how I defined the macro, in the meantime I'll close this. Let me know if you have anymore problems, best of luck.

closed 2022-08-04 by cujomalainey