cujomalainey/antplus-arduino #35: Seeeduino nRF52840 problems

HristoMarkow, 2022-11-29:
Hi,

I've downloaded the project and tried to compile the example "Bicycle Power Sensor",
but first it throws an error on line 19
`C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved20221029-18884-13azw4e.zxko\BicyclePowerSensor\BicyclePowerSensor.ino: In function 'void setup()':
C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved20221029-18884-13azw4e.zxko\BicyclePowerSensor\BicyclePowerSensor.ino:19:19: error: 'Serial2' was not declared in this scope; did you mean 'Serial'?
   19 | #define antSerial Serial2
      |                   ^~~~~~~
C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved20221029-18884-13azw4e.zxko\BicyclePowerSensor\BicyclePowerSensor.ino:61:5: note: in expansion of macro 'antSerial'
   61 |     antSerial.begin(BAUD_RATE);
      |     ^~~~~~~~~

exit status 1

Compilation error: 'Serial2' was not declared in this scope; did you mean 'Serial'?`

then I've changed to Serial1 and put the network key from thisisant.com, the sketch is compiled and uploaded,
but my Edge 1040 didn't find the sensor.

Is it possible because the device is new?

Regards,
Hristo

cujomalainey, 2022-11-29:
Hi Hristo,

Sounds like you used a serial driver when you meant to use the softdevice driver. The serial driver is meant to control a D52 ANT chip external to the board you are programming. See this [example](https://github.com/cujomalainey/ant-arduino/blob/master/examples/NativeAnt/NativeAnt.ino) for how to use the softdevice driver. You can swap it for the serial driver in the router.  You need to make sure you have a ANT compatible [softdevice](https://www.thisisant.com/developer/components/nrf52832/) loaded onto your device in order for it to work. Here is a [blog post](https://blogarak.wordpress.com/2020/03/15/s340-softdevice-adafruit-nrf52840-express-feather/) on how to do it for the Adafruit Feather Express, it will likely be a different procedure for the Seeeduino. Let me know if you have any more questions.

HristoMarkow, 2022-11-30:
Hi Curtis,

when I try to compile/verify it gives
`C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved20221030-18884-2ur1xa.97di\sketch_nov30a\sketch_nov30a.ino:31:1: error: 'ArduinoNativeAnt' does not name a type; did you mean 'ArduinoSerialAnt'?
 ..............`

Also tried with suggested change from Arduino IDE but there is again a compilation problem :)
` from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved20221030-18884-2ur1xa.97di\sketch_nov30a\sketch_nov30a.ino:19:
c:\Users\User\Documents\Arduino\libraries\ANT-Arduino\src/BaseClasses/ANT_BaseSerialAnt.h:49:10: note: candidate: 'void BaseSerialAnt<T>::begin(T&) [with T = Stream]'
   49 |     void begin(T &serial);
      |          ^~~~~
c:\Users\User\Documents\Arduino\libraries\ANT-Arduino\src/BaseClasses/ANT_BaseSerialAnt.h:49:10: note:   candidate expects 1 argument, 2 provided

exit status 1

Compilation error: no matching function for call to 'ArduinoSerialAnt::begin(int, int)'`

my target is to make a controller that can "talk" with Garmin and commands some chinese lights.
I already got Bontrager Ion Pro RT/Flare RT Light Set.

The dilemma: is this board working or not?!?

Regards,
Hristo

cujomalainey, 2022-11-30:
Hi Hiristo,

The NativeAnt driver is only available when the respective build flag is exposed for the correct soft device. E.g. if `S340` is defined then the driver is available. The build system should expose it if configured properly. If you don't have the right softdevice then you should be able to call functions in ant_interface.h have you verified you have the correct soft device loaded?

HristoMarkow, 2022-12-01:
Hi Curtis,

I suppose you talk about the menu Tools / SoftDevice which is shown when I choose
[image]
but when I choose my exact model of board this item from the menu disappears /the snapshot from the second picture/

this information is located in file INFO_UF2.TXT into folder E: when I plug the board via cable on my Windows 10 computer
**UF2 Bootloader 0.6.2-12-g459adc9-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Seeed XIAO nRF52840
Board-ID: Seeed_XIAO_nRF52840_Sense
Date: Nov 30 2021
SoftDevice: S140 7.3.0**

according the AliExpress it should be "Seeeduino XIAO BLE SENSE Development Board Module BLE 5.0 nRF52840 Nano/uno Arm Microcontroller for Arduino"

cujomalainey, 2022-12-01:
There is your issue, the S140 is BLE only softdevice, you have to swap that out for a ANT enabled device (see link in my first comment) without that you won't be able to use your device for ANT. How that is done on this particular device I do not know, on adafruit boards the softdevice is bundled with the bootloader. This is something you will have to investigate or swap your board for something that has already been explored by other.

HristoMarkow, 2022-12-05:
Hi Curtis,

Last question: seems I have to fix my Seeeduino or to use another product, so is there a Arduino board based on nRF51 /I read for nRF51422/ that is ready to "work" w/o flashing?

Regards,
Hristo

cujomalainey, 2022-12-05:
If you plan to have the project be a single microcontroller with arduino, then no. The only boards that comes pre-flashed with ANT is the dynastream dev boards (D52, N5, etc.) which are not Arduino ready. The only way to get around the flashing is to have a D52 be a separate board and interface with ANT via the serial interface on the D52 as it comes preloaded with the "network processor" (See my serial driver).

Also side note, it turns out all the guides we wrote ~2 years ago are [out of date/useless](https://github.com/cujomalainey/ant-arduino/issues/56#issuecomment-1336488938) as adafruit has completely modified the softdevice process for the board we tested on.

HristoMarkow, 2022-12-07:
Got it, seems a USB flash from Garmin with some Raspberry board will do the job because
I assume that the license is included in price of the flash drive itself.
[image]

As far I understand nothing available on the market based over nRF51 or nRF52 will be useful.

cujomalainey, 2022-12-08:
yes, you don't need to worry about the license on a USB device, its simply the serial protocol over USB. I'm currently working on a Rust implementation that includes a USB interface. Might be able to release it in a month or so (copyright nonsense is forcing it private right now)

HristoMarkow, 2022-12-19:
In short: I can use arduino board but should plug the USB adapter, right?

cujomalainey, 2022-12-19:
If you have a board that can be a USB host, I believe so, never tried it myself though

cujomalainey, 2022-12-28:
Closing due to lack of activity, please reopen if you have any more questions

closed 2022-12-28 by cujomalainey