cujomalainey/antplus-arduino #36: Ant+ FE-C sensor profile support

Pilleo, 2023-04-09:
Hi, I am trying to stream data to my watch. Already failed with ble ftms because garmin does not support that profile, now I am hoping for ant+. I need FE-C sensor lib because I have an old crosstrainer that is measuring a lot of advanced data, but it is useless as it does not let to store it in any way. So having fe-c sensor for arduino would be super nice, cause my plan is make it compatible with my watch.
Also I am considering to use Rp zero if that will be easier, but have not find any lib for any language to stream FE-C. Maybe somebody knows such a lib?
Thank you

cujomalainey, 2023-04-10:
Hi @Pilleo 

I won't have time to a FEC implementation as I am spending all my spare time on the rust version of this lib. Should you make one I will welcome the contribution as you are not the only one interested in the profile.

Regarding library, and RP. The only library I can think of that would even partially implement the profile is the .Net library for Simulant and I believe that would just give you datapage data types. And I don't believe that is cross compiled for the .net core so I don't think it works on anything other than windows.  On the other hand, my rust library ([ant-rs](https://github.com/cujomalainey/ant-rs)) will work with a raspberry Pi with a usb radio plugged in, but it is still very early development and I won't be implementing FEC for a while since I am working on a bunch of core design principals before I publish. That being said, the library is functional and you could add FEC now if you wanted.

Pilleo, 2023-04-11:
Thank you @cujomalainey, I was looking into your code before writing and it seemed like way over my head with experience only in java. Just is awesome, btw:) How long do you think it would take you to implement it? I this multiplying it by 15 could give an estimation for me, so I want to know it before I start developing anything.

cujomalainey, 2023-04-11:
No problem, its actually much easier to do than the C++ code in antplus-arduino. Couple of points I recommend.
* Use the development branch
* Most (like 3/4) of the work is just copying the datapages out of the datasheet
  * Copying test values from Simulant+ into unit tests makes testing and debugging byte parameter ranges very easy
* if you are just looking to receive data then you are golden
  * ANT-FS and bursting for file transfers is completely untested, not even sure is 25% implemented
* A lot of my work right now is around the [MessageHandler](https://github.com/cujomalainey/ant-rs/blob/development/ant/src/plus/common/helpers.rs#L48) which you don't need, its just a common channel configuration tool, that I can add later. So just hardcode configuring your channel to avoid that mess
* TX is a bit more work, I am trying to make the library very easy to work with minimal ownership messes. As a result unlike antplus-arduino where a profile can transmit whenever it wants, profiles have to be asked if they have anything to send. The downside is profiles have to maintain all un-transmitted state until their next transmission window.

Honestly if you commit the datapages I could probably hack together a profile pretty quickly. The most annoying part is the fact that [packing lib](https://docs.rs/packed_struct/latest/packed_struct/) its bit endianess is a bit finicky. [Datapages](https://github.com/cujomalainey/ant-rs/blob/development/ant/src/plus/profiles/heart_rate/datapages.rs#L71) are even easier because they are fixed length unlike radio messages.

Another possible method of attack you can do is reuse the antplus-arduino code, and just mock a arduino serial driver against a usb device and reuse the [antplus-arduino](https://github.com/cujomalainey/antplus-arduino/tree/master/src/Profiles/Fec) code not on an arduino

Pilleo, 2023-04-12:
Wow, a lot of options to consider:) I am making a sensor for my crosstrainer, so I will need to write data. 
I could definitely do some copy-paste coding if that helps you with fe-c, and tbh this is the most clear option to me. I will contact you with mail

cujomalainey, 2023-07-17:
Closing out as I believe this has been resolved. Please reopen if this has been closed in error. Thanks

closed 2023-07-17 by cujomalainey