cujomalainey/antplus-arduino #37: Unable to get example working - seeeduino nrf52840

Timee2023, 2023-06-23:
Hello,

I bought a seeeduino nrf52840 and wanted to try the heart rate display. I uploaded the code to the chip and installed both the libraries mentioned (ant-arduino and antplus-arduino). I believe I got the right network key from thisisant.com (The ANT+ network key) and changed it for the default placeholder in heart rate display code and uploaded that. On the serial i just see "Running" and nothing else. Was I meant to change anything else. Quite new to this.

Thanks for any help.

cujomalainey, 2023-06-23:
I should really label these examples better as you are not the first to be confused.

"Serial" examples are meant for devices connected to an external nRF52 loaded with the network processor firmware. You are looking for the "native" examples if you are running your program on the nRF52. You will need to manually install the ANT headers into the SDK as I cannot ship them. Also make sure your chip is loaded with an ANT compatible softdevice.

Timee2023, 2023-06-23:
Thanks for the extra information.

Timee2023, 2023-09-30:
Hi @cujomalainey finally coming back to this as got some free time. Wondering if I could get some assistance, not sure if this idea works.

I have a GARMIN ANT+ USB Stick.
I have a USB to TTL converter, image below: 

[image]

I desoldered the male head and soldered on a female head on the converter. This allowed me to connect the ANT+ stick to the converter. I proceeded to wire up RX and TX from the converter to the inverse position on the Seeeduino NRF52840 to attempt to get data from the stick.

I'm not sure whether this idea was going to work as the ANT+ stick may require a host. My idea stemmed from the fact that this ANT+ stick has a NRF24AP2 inside, effectively the same as the [SPARKFUN NRF24AP1](https://www.sparkfun.com/products/retired/8565) as seen below: 

[image]

This chip has the embedded ANT protocol inside of it.

Is there anyway I could potentially use the USB ANT+ stick if my setup was wrong? Additionally, I guess the setup with the SPARKFUN board would work as the ANT protocol is embedded inside of it.

Any help or suggestions are appreciated, thanks!

cujomalainey, 2023-10-01:
@Timee2023 indeed the ANT stick is valid for the serial protocol, its what I use actually for my rust development. The problem is I don't work in embedded USB so I am not sure about your setup if it will work on not. I know there are special parts of usb like host/slave.

My recommendation is lookup the chip on the USB to TTY converter and see if it can do what you are looking for. I know the ANT stick is essentially serial over usb and is actually automatically converted to a serial device on linux so at worst you are likely not far off from a workable solution.

Timee2023, 2023-10-04:
Will keep you updated, thanks

closed 2023-10-04 by Timee2023